pyxem/pyxem #1096: Post 0.19.0 tidy ups

pc494, 2024-06-22:
Hello, apologies for my absence, it sounds like things went well at NTNU.

I've got a little bit of time over the coming weeks so I'm going to go back to some of my tidying. In the short term this means:

- [ ] deleting all code that was marked for removal in the `0.19.0` series
- [ ] moving all other deprecations up to `0.20.0`
- [ ] picking up any outstanding reviews I have here/orix/diffsims (please ping me)

CSSFrancis, 2024-06-22:
@pc494 no worries! Any time you have is greatly appreciated :) the second half of this week I should have some time to work on some things as well.

I guess the question is do we want to make a 0.20.0 or go all in for a 1.0.0 release. I feel like the big refactors are done now and there isn't anything else I would like changed other than maybe some better examples on calibration.

pc494, 2024-06-24:
I think we probably go for a `1.0.0` release, but it also can't hurt to bring the deprecations forward as a safety measure

pc494, 2024-08-07:
We're pushing on for `1.0.0`, closing :)

closed 2024-08-07 by pc494