pyxem/pyxem #1097: Saving `BeamShift` signal from `get_direct_beam_position`: metadata error

magnunor, 2024-07-04:
In the current `main` branch, saving `BeamShift` signals as `.hspy` files generated from `get_direct_beam_position` is not possible. This is due to the `s_bs.metadata.Shifts.signal_axes` containing the `Object` dtype: `TypeError: Object dtype dtype('O') has no native HDF5 equivalent`.

-------------

To replicate:

```python
import pyxem as pxm
s = pxm.data.dummy_data.get_holz_simple_test_signal()
s_bs = s.get_direct_beam_position(method="center_of_mass")
s_bs.save("test0.hspy") # Gives error
del s_bs.metadata.Shifts
s_bs.save("test1.hspy") # Does not give error
```

CSSFrancis, 2024-08-06:
@magnunor This should have been fixed in https://github.com/hyperspy/hyperspy/pull/3342?

Can you make sure that you are using 2.1.0 otherwise I can look into this further.

magnunor, 2024-08-08:
I tested this with Hyperspy 2.1.1, and I still get the error: `TypeError: Object dtype dtype('O') has no native HDF5 equivalent`

CSSFrancis, 2024-08-08:
A more miniumum example.  

```python

import hyperspy.api as hs
from hyperspy.axes import UniformDataAxis
import numpy as np
s = hs.signals.Signal1D(np.arange(10))
axis = UniformDataAxis(name="x", units="ly", size=10)
s.metadata.set_item("axis", (axis,))
s.save("temp.hspy")
hs.load("temp.hspy").metadata
```