pyxem/pyxem #1098: Installation via pip fails

opens21, 2024-07-16:
**Describe the bug**
The package is based on an old pyfai version of 2023.9.0 which fails to build. How can I install pyxem if the pyfai build fails?
```
____________________________________________________
pyFAI v 2023.09 build fails with those dependencies: [https://github.com/silx-kit/pyFAI/issues/2232]

    Python 3.12
    gcc14
    cython 3.0.10
    numpy 2
____________________________________________________

```

CSSFrancis, 2024-07-16:
Hi @opens21 Thank you for raising the issue!

You might need to try installing using python 3.11 rather than python 3.12, we don't currently test against python 3.12 (I'll ty to add that later this week). Let me know if you can't do that for some reason!   We are planning to remove pyfai as a dependancy (partially because of this issue) and I can try to go ahead with that today and make a patch release if that is something you need right now otherwise it might be a week before I get to it. 

Otherwise the [Hyperspy Bundle](https://hyperspy.org/hyperspy-bundle/) is always a good place to start if you are having issues setting up an environment.

ericpre, 2024-07-16:
You will need to use numpy < 2.0, because there is another dependency (`numpy-quaternion` through `orix`) which is not compatible with numpy 2, see https://github.com/moble/quaternion/issues/229.

opens21, 2024-07-16:
@CSSFrancis It is not super-urgent, but would be nice if I can get it working in ~1 week from now.

@ericpre Does it mean I have to uninstall numpy 2 or is there a workaround, where I can have numpy 1 and numpy 2?

ericpre, 2024-07-17:
You can't have numpy 1 and numpy 2 installed in the same environment.

CSSFrancis, 2024-07-17:
@opens21 Can you maybe give some context as to why you want to use Python 3.12 and numpy 2.0.

Both are things we intend to support in the future. Python 3.12 is fairly doable, but numpy 2.0.0 is dependent on lots of other packages which we have less control over. In general I'd advise against Python 3.12 for at least a little bit. Support for it among packages has been slow mostly because Python 3.13 has a lot of the exciting things people are waiting for. (For example even some packages supported by Google don't support it)

I'd recommend using [mamba](https://mamba.readthedocs.io/en/latest/installation/micromamba-installation.html) instead of pip to manage Python packages and from there you can specifically make a python 3.11 environment.

You can also just install the bundle as well. That should always work and will give you a complete Python environment with many of the packages you might be interested in. I'd highly recommend that if installing everything is something you are struggling with.

closed 2024-12-19 by CSSFrancis