pyxem/pyxem #1099: Installation stop gap for numpy 2.0.0

pc494, 2024-07-23:
Related to #1098 

At present, pyxem has too many dependencies that will not play nicely with numpy 2. In the short term I think we should pin ourselves to 1.x series numpy and write a very large installation warning at the bottom saying words to the effect of:

> If you do not create a clean environment attempting to install pyxem is very likely to cause you problems.

I could probably implement this for a patch over the next week? @CSSFrancis, @ericpre what are your thoughts?

pc494, 2024-07-23:
Okay, I've just dug into where Hyperspy have got to, and I get the feeling we're going to be rather unpopular if we pin ourselves to numpy < 2

Is `quaternion` our only known blocker?

ericpre, 2024-07-23:
Maybe use https://github.com/moble/quaternionic (pure python and accelerated with numba) to replace quaternion? Support for numpy 2 in quaternion seems to be difficult: https://github.com/moble/quaternion/issues/229.

CSSFrancis, 2024-07-24:
I think @ericpre is correct.  We probably want to replace quaternion with quaternionic and then rip out `pyfai` and make a new release.  I think being a sticking point for a numpy 2.0.0 release is going to make us unpopular. 

I have time next week to do this if someone else doesn't get to it first.

pc494, 2024-07-24:
> Maybe use https://github.com/moble/quaternionic (pure python and accelerated with numba) to replace quaternion? Support for numpy 2 in quaternion seems to be difficult: [moble/quaternion#229](https://github.com/moble/quaternion/issues/229).

This seems like it would be the correct approach, but it may not be a trivial migration and orix is very short on dev resourcing.

opens21, 2024-08-14:
Any idea if this is going to happen soon? Do you have any timescale?

CSSFrancis, 2024-08-14:
@opens21 can you try using python 3.11? 

This depends on some other packages and we have to wait for them to update.

hakonanes, 2024-10-05:
orix v0.13.2 is now compatible with NumPy v1  and v2.

hakonanes, 2024-10-08:
Seems like latest diffsims v0.6.0 is not compatible with NumPy v2: https://github.com/pyxem/diffsims/issues/227.

closed 2024-12-19 by CSSFrancis