pyxem/pyxem #1100: Repetitive indexing changes the input signal?

OverEthanol, 2024-07-24:
Hi, I am not sure if this is a bug or an intended feature. Suppose that I have loaded and processed some TEM diffraction pattern as variable `s_pol_gamma` and it looks like this (done with `s_pol_gamma.plot(gamma=0.1)` and all the following plots are plotted in the same way):

[image]

Then I performed the indexing by 

```Python
result = s_pol_gamma.get_orientation(simulation=sim, n_best=10, frac_keep=1.0)
```

and when I plot out `s_pol_gamma` again, now it looks like this

[image]

and if I do it again, then

[image]

and again...

[image]

It seems like the electron counts values just keep dropping down the more repetitive indexing I do on the same signal. Is this because each time when indexing is performed, it takes away the pixel values that are matching...??? I don't know understand :D

pc494, 2024-07-29:
I'm going to label this as a bug for now, I'm not sure we have the resources to look into it though :(

CSSFrancis, 2024-08-05:
@OverEthanol I think I have some time today to look into this.  Hopefully we can get a bug fix out rather quickly!

CSSFrancis, 2024-08-06:
@OverEthanol What version of pyxem do you have?  This was a bug in 0.19.0 but should have been fixed in 0.19.1!

closed 2025-05-27 by CSSFrancis