pyxem/pyxem #1106: center_direct_beam function

opens21, 2024-09-01:
**Describe the bug**
I can correctly center the direct beam, but the subsequent data processing steps are not documented. The plot of the corrected center beam does not account for the input vmin and vmax. Additionally, it displays the neglected diffraction region after centering; however, I want to plot it without this region while navigating the 4D dataset. The Centering the Zero Beam example results in an error, as s_shifts.get_linear_plane fails in the tutorial.

How can I get the data which only includes the common reciprocal space which is present for all x and y coordinates?

**To Reproduce**
```
import pyxem as pxm
dp = pxm.signals.ElectronDiffraction2D(s4D)
dp_new = dp.center_direct_beam(method='interpolate', upsample_factor=20, sigma=0.2, kind='linear', inplace=False)
plot_s4D(dp, norm='log', cmap = 'inferno', scalebar=False, title ='', aspect='auto', vmin = 0.2, vmax = 1e2)
dp_new.plot(norm='log', cmap = 'inferno', scalebar=False, title ='', aspect='auto', vmin = 0.2, vmax = 1e2)
```

[image]

CSSFrancis, 2024-09-02:
@opens21 

Thanks for raising this issue! I can take a look at this tomorrow and improve the documentation a little bit :) Foodback like this is always appreciated.

One thing that might help (and I like to do) is set vmax="96th" etc  which will set the maximum value equal to the 96th percentile. 

Otherwise you can manually slice the signal using `s.isig[10:-10, 10:-10]` to slice the outside pixels related to the shift.

The tutorials... should probably be removed. They are hard to maintain and the Examples are usually much more up to date. Although full workflows are probably nice.  I'll put it on my list to fix those this week.

CSSFrancis, 2024-09-04:
Just also a point that `norm="symlog"` should work better for data with 0 values.

CSSFrancis, 2024-09-06:
@opens21 Can I close this?