pyxem/pyxem #1107: Going down to one form of "examples of how to do things"

pc494, 2024-09-02:
Building on the thoughts expressed in

_Originally posted by @CSSFrancis in https://github.com/pyxem/pyxem/issues/1106#issuecomment-2324717110_

Having two, similar, but no identical sets of "Tutorials" creates a lot of maintenance issues for a basically unstaffed team. While the notebook approach works in other pyxem projects, I think right now pyxem (this repo) has better Readthedocs style documentation meaning the repo "pyxem-demos" should go. 

@CSSFrancis does this seem fair to you?

CSSFrancis, 2024-09-03:
@pc494 im okay with archiving the demos and then removing them from the documentation.

I'm up for suggestions. One option would be to make a gallery of tutorial repos based on the different tutorials released and tag them with a version.

The right place to do this would be upstream with Hyperspy.The main problem is that we have a bunch of repos from:

- Papers 
- Tutorials 
- Random snippets of code

It would be fairly easy to:

- Tag them with a version of pyxem, hyperspy etc.
- Include a link to the dataset/datasets
- Indicate if they should be run on every new minor release. 

Then we could make a list with all of the resources and the versions they need.

CSSFrancis, 2024-09-03:
@ericpre any thoughts here?

I might add this to the hyperspy website.

This is also related:

https://github.com/hyperspy/exspy/pull/49

pc494, 2024-09-03:
I suppose my main opinion is that this is in fact a fairly large piece of work. Might be worth you and I (@CSSFrancis) sitting down to chat about it (e.g. our personal capacities).

CSSFrancis, 2024-09-03:
@pc494 Sure I would love to sit down and talk about it!  I can send you an email :)

TimothyLambdenCambridge, 2024-09-24:
@pc494 @CSSFrancis 

I agree that focussing on one set of tutorials would be benefical and in this case would be worth archiving the pyxem-demos with a link to newer/better documentsed tutorials where applicable. Less sure on `nbsphinx `documentation but can learn :) and more than happy to help!

I guess they can be separated based on the ones that are more robust to minor changes stay on the read-docs section e.g. examples, and those that aren't get archived e.g. random snippets of code?

TimothyLambdenCambridge, 2024-09-24:
Also related: 

https://github.com/pyxem/pyxem-demos/issues/95

p-valiant, 2025-06-26:
Is there something like demo 09 in the examples? I noticed that the demos have issues with version 0.20 and above, and the notebooks from the hyperspy workshop this year don't seem to include an equivalent.

CSSFrancis, 2025-06-30:
@p-valiant I had a little typo.  It's cfrancis at directelectron dot com