pyxem/pyxem #1115: Deprecate `make_probe_navigator`?

ericpre, 2024-09-16:
When reviewing https://github.com/pyxem/pyxem/pull/1108, I wondered if the [`make_probe_navigation`](https://github.com/pyxem/pyxem/blob/971a7be372fe4b6f5f73b95833448e67512911c0/pyxem/signals/diffraction2d.py#L1372C9-L1372C30) method should be deprecated in favour of `compute_navigator`?

pc494, 2024-09-17:
Seems sensible, probably more @CSSFrancis's domain than mine though.

ericpre, 2024-09-17:
One significance difference is that `compute_navigator` is only for lazy signals, while `make_probe_navigation` is also used for non-lazy signals. Maybe it is worth implementing `compute_navigator` for non-lazy signals in hyperspy before deprecating it - see https://github.com/hyperspy/hyperspy/issues/2639.