pyxem/pyxem #1117: v0.20.0 might be a good idea

pc494, 2024-09-20:
v20.0.0 would mainly serve to make use of the fact that with #1103 & orix v0.13.1 (orix-base) we would (in theory) be numpy 2.0.0 compatible. This would help our install process (see #1098 and #1099) and basically comes for free (all the code is on the RELEASE_NEXT_MAJOR branch).

@CSSFrancis thoughts?

CSSFrancis, 2024-09-20:
@pc494 Sure :) We should merge #1092 as well then :)

CSSFrancis, 2024-09-20:
I made a milestone for it and will go through and make sure all of the code is ready to go :)

hakonanes, 2024-09-20:
Turns out latest diffpy.structure is not compatible with NumPy 2.0 (https://github.com/diffpy/diffpy.structure/issues/68), so orix 0.13.1 is neither. I've reached out to hear what it would take to make diffpy.structure compatible.

pc494, 2024-09-20:
> Turns out latest diffpy.structure is not compatible with NumPy 2.0 ([diffpy/diffpy.structure#68](https://github.com/diffpy/diffpy.structure/issues/68)), so orix 0.13.1 is neither. I've reached out to hear what it would take to make diffpy.structure compatible.

This is disappointing, but we're doing what we can.

pc494, 2024-09-24:
Okay, it now looks like orix (see https://github.com/pyxem/orix/issues/506) should be numpy1 + numpy2 compatible in the next week or so. 

Should we try and pick up everything on the milestone fairly soon after that?

CSSFrancis, 2024-10-25:
Okay so we are now numpy 2.0.0 Compliant:

Scoping out some changes that should be in before 0.20.0

- [x] #1121 --> Done in #1122 
- [x] #1092 
- [x] #1108 
- [ ] #1116
- [ ]  #1120
- [ ]

ericpre, 2024-11-15:
@CSSFrancis, do you a date for the release adding support for numpy 2.0? I am asking to know if this is worth to wait for it to make a release of the hyperspy bundle.

CSSFrancis, 2024-11-15:
Let me try to do that today!

closed 2024-12-09 by CSSFrancis