pyxem/pyxem #1121: Version Switcher Broken

CSSFrancis, 2024-10-24:
**Describe the bug**
The Version switcher seems to be broken.  The _static/switcher.json page seems unreachable? For some reason.

Let me test a couple of other things.

ericpre, 2024-10-25:
Is it possible that readthedocs changed its configuration and the content of `_static` folder is not copied to the output anymore?
Maybe this line needs to be uncommented to make sure this `json` file is present in the output?
https://github.com/pyxem/pyxem/blob/7446071464777281c0068741b61b19fbe4d188b2/doc/conf.py#L129

CSSFrancis, 2024-10-25:
> Is it possible that readthedocs changed its configuration and the content of `_static` folder is not copied to the output anymore? Maybe this line needs to be uncommented to make sure this `json` [...]

Hmm that's a good point.  Let me check that!

closed 2024-10-25 by CSSFrancis