pyxem/pyxem #1127: Orientation mapping fails when `n_best` exceeds the azimuthal sampling and non-intuitive axes for template matching results.

emichr, 2024-11-19:
**Describe the bug**
When template matching a small (3,3|256,256) dataset azimuthally integrated to size (3,3|360, 181) with all the 1081 templates in a bank (`n_best=simulations.current_size`), the following exception is raised: `ValueError: could not broadcast input array from shape (108,4) into shape (1081,4)`. See traceback at the end. 

It looks to me that the results are attempted to be cast into the shape of the azimuthally integrated data, rather than a suitably sized output. Looking at the results from orientation mapping with fewer `n_best` shows also a strange axes_manager:
[image]

The fact that the template matching results are stored in axes "copied" from the signal that is template matched is not intuitive.

I am not entirely sure if the error lies somewhere in my own code, pyxem, or hyperspy, but any help is greatly appreciated!

**To Reproduce**
See the [MWE.zip](https://github.com/user-attachments/files/17815017/MWE.zip) for required files and an MWE jupyter notebook. The notebook is based on [this notebook](https://github.com/TEM-Gemini-Centre/nordtemhub_4DSTEMworkshop_2024/blob/main/Orientation_mapping_SPED-Ag/Orientation_mapping_SPED-Ag.ipynb) that was used in the NordTEMHub workshop this summer. 

The code is included here as well in a (semi) MWE:
# MWE

## import

```python
%matplotlib widget
import hyperspy.api as hs
import pyxem as pxm
import numpy as np

from diffsims.generators.simulation_generator import SimulationGenerator

from orix.quaternion import Orientation
from orix.crystal_map import Phase
from orix import plot
from orix.sampling import get_sample_reduced_fundamental
```

## Load

```python
signal = hs.load('subset.hspy')
```

## Load cif file and create beam directions

```python
Ag = Phase.from_cif('Ag.cif')

#Create beam orientations to simulate
resolution = 1.0 #The angular spacing between each orientation
beam_directions = get_sample_reduced_fundamental(resolution, point_group=Ag.point_group) #Get a uniform sampling of euler angles
print(f'Simulated {beam_directions.size} beam directions')
beam_orientations = Orientation(beam_directions, symmetry=Ag.point_group)
beam_orientations.scatter('ipf')
```

## Create simulations

```python
# Create simulations
minimum_intensity = 1E-20 #Tuneable parameter
max_excitation_error = 1E-1 #Tuneable parameter

sim_gen = SimulationGenerator(
    accelerating_voltage=signal.metadata.Acquisition_instrument.TEM.beam_energy,
    scattering_params='lobato',
    shape_factor_model='linear',
    minimum_intensity=minimum_intensity
) #Create a diffraction simulation generator

calibration = signal.axes_manager[-1].scale #Get the calibration from the signal
reciprocal_radius = np.max(np.abs(signal.axes_manager[-1].axis)) #Get the maximum reciprocal radius to simulate
half_shape = np.min(signal.axes_manager.signal_shape)//2 #Get the half-shape of the patterns

simulations = sim_gen.calculate_diffraction2d(
[... 96 lines cut ...]
    353     """Compute this dask collection
    354 
    355     This turns a lazy Dask collection into its in-memory equivalent.
   (...)
    374     dask.compute
    375     """
--> 376     (result,) = compute(self, traverse=False, **kwargs)
    377     return result

File [~\AppData\Local\miniforge3\envs\pyxem0.19.1\Lib\site-packages\dask\base.py:662](http://localhost:8888/lab/tree/LocalDocuments/GitHub/developments/Templatematching/~/AppData/Local/miniforge3/envs/pyxem0.19.1/Lib/site-packages/dask/base.py#line=661), in compute(traverse, optimize_graph, scheduler, get, *args, **kwargs)
    659     postcomputes.append(x.__dask_postcompute__())
    661 with shorten_traceback():
--> 662     results = schedule(dsk, keys, **kwargs)
    664 return repack([f(r, *a) for r, (f, a) in zip(results, postcomputes)])

File [~\AppData\Local\miniforge3\envs\pyxem0.19.1\Lib\site-packages\hyperspy\misc\utils.py:1360](http://localhost:8888/lab/tree/LocalDocuments/GitHub/developments/Templatematching/~/AppData/Local/miniforge3/envs/pyxem0.19.1/Lib/site-packages/hyperspy/misc/utils.py#line=1359), in process_function_blockwise(data, function, nav_indexes, output_signal_size, output_dtype, arg_keys, *args, **kwargs)
   1358     for nav_index in np.ndindex(chunk_nav_shape):
   1359         islice = np.s_[nav_index]
-> 1360         output_array[islice] = function(data[islice], **kwargs)
   1361 else:
   1362     # There are BaseSignals which iterate alongside the data
   1363     for index in np.ndindex(chunk_nav_shape):

ValueError: could not broadcast input array from shape (108,4) into shape (1081,4)
```

emichr, 2024-11-19:
It seems that including `frac_keep=1.0` solves the issue.

The point with non-intuitive axes for the results still remain however, but that might be better put in a separate issue. I'll close this for now.

closed 2024-11-19 by emichr

CSSFrancis, 2024-11-19:
> It seems that including `frac_keep=1.0` solves the issue.
> 
> The point with non-intuitive axes for the results still remain however, but that might be better put in a separate issue. I'll close this for now.

I'll move this to a disucssion :)   It's good to have some Q/A there for people to use as reference.