pyxem/pyxem #1134: Azimuthal integration - Artefact in radial profile

markaturner04, 2025-02-06:
I have been performing azimuthal integration of 4D STEM data for the purpose of ePDF analysis. However, I have encountered a discontinuity at large scattering vector shown here, due to what I guess is that at the corners of the diffraction frame, there are less bins being counted and so a sharp drop off in intensity. 

[image]

Is there a way that instead an azimuthal average could be computed so that the count is normalised with respect to the number of contributing bins, or if anyone has any direction on how to do this?

CSSFrancis, 2025-02-06:
@markaturner04 This is a great question!  I guess the answer is that there are a couple of ways to do this although it might be best to think about what the best one is.  

In the following code the `mean=True` parameter will normalize vs the number of pixels in each bin.  I'm not sure I spent a lot of time thinking about the edges as I kind of assumed that most people would just crop them off.   

```python 

import pyxem
import hyperspy.api as hs
s = pyxem.data.pdnip_glass()
s.calibration.center =None # Use direct center of diffraction pattern
integral = s.get_azimuthal_integral1d(100, mean=True)
integral_unnormalized = s.get_azimuthal_integral1d(100, mean=False)

hs.plot.plot_spectra((integral.sum(), integral_unnormalized.sum()),legend=("unnormalized", "normalized"), style="mosaic")
```

[image]

Maybe we can test this a little bit. 

```python

import numpy as np
padded_array = s.map(np.pad, pad_width =2, inplace=False)
padded_array.calibration.center =None # Use direct center of diffraction pattern
integral = s.get_azimuthal_integral1d(100, mean=True)
mask = np.zeros(s.axes_manager.signal_shape, dtype=bool)
mask = np.pad(mask, pad_width=2, constant_values=1)
padded_array.calibration.mask = mask

integral_padded = padded_array.get_azimuthal_integral2d(100, mean=True)
hs.plot.plot_spectra((integral.sum(), integral_padded.nansum(axis=(0,1,2))/360), 
                     legend=("Integral", "Masked Integral"))
```

There is some weirdness going on but the bllip is still there... 

[image]

It's hard to know what exactly is going on there.. It might be calculation error?  One possibility is that it's a misalignment between the pixel extent and the integration box where a half pixel shift might need to be introduced... I thought that the mask would remove it but for some reason it seems to be fairly consistent.

CSSFrancis, 2025-02-06:
I guess for completeness here is the centering code with a beam stop... Although it doesn't seem to change things very much....

```python
import pyxem as pxm
s = pyxem.data.pdnip_glass()
sum_dp = s.sum()

mask = sum_dp.get_direct_beam_mask(radius=20)

mask.isig[:2,:]=1
mask.isig[:,-2:]=1

center, affine, params, pos = pxm.utils.ransac_ellipse_tools.determine_ellipse(
    sum_dp,
    return_params=True,
    mask=mask.data,
    num_points=1500,
    use_ransac=False,
)
el, in_points = pxm.utils.ransac_ellipse_tools.ellipse_to_markers(
    ellipse_array=params,
    points=pos,
)

sum_dp.calibration.scale=1
sum_dp.calibration.center=[0,0]

sum_dp.plot()
sum_dp.add_marker(in_points, plot_marker=True)
sum_dp.add_marker(el, plot_marker=True)

s.calibration.center = center[::-1]

```

markaturner04, 2025-02-06:
Thank you very much for the response, using the mean = True argument does, like you show, improve the resulting profile and reduce the discontinuity somewhat. This blip in the resulting profile is still there, like you showed and does propagate to the PDF analysis that I am doing. I wonder if this can't be avoided with the integration step, whether the blip can be removed through smoothing the profile at this point in reciprocal space. Maybe somewhat similar to the remove spikes function?

CSSFrancis, 2025-02-07:
@markaturner04 I guess the first thing to think about is do you need to use those pixels?  Your're not uniformly sampling the entire space anyways so I'd hesitate to use that region of the diffraction patterns anyways.  

Is there a reason to use them or can you just set the `radial_range` or crop the dataset somehow?

ericpre, 2025-02-07:
Since the data on the edge are missing, would it make sense to exclude the corners? It would also have the advantage to make the computation faster.

The docstring mentioned that it is possible to specify `azimuthal_range` but from a quick screening at the code, it doesn't seem to be the case for `get_azimuthal_integral1d`!

CSSFrancis, 2025-02-09:
@viljarjf Good test... 

Not super helpful but at least it seems like the mean calculation is working as intended 😅 

```python
from pyxem.signals import ElectronDiffraction2D
import numpy as np

data = np.ones((100, 100)) # Replace this with data from e.g. a tif file
sig = ElectronDiffraction2D(data)
sig.calibration.center = None # Sets polar unwrapping center to the middle of the image
profile = sig.get_azimuthal_integral1d(npt=100, mean=True)

profile.plot()
```

[image]

CSSFrancis, 2025-02-20:
@markaturner04 The graph you are seeing there is (basically) the same.  If you look at the actual data

`print(s.data)` you get something like 

```
array([0.99999992, 0.99999997, 0.99999998, 0.99999999, 0.99999999,
0.99999999, 0.99999999, 0.99999999, 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        , 1.        , 1.        , 1.        , 0.99999999,
       0.99999999, 0.99999999, 0.99999999, 0.99999998, 0.99999994])
```

The reason for the change from 19.1 to 20.0 is that we added in a small eps value becuase ohterwise it was possible to divide by 0.  For some bins with very few pixels, however, this seems to cause some issues.

closed 2025-04-24 by CSSFrancis