pyxem/pyxem #1135: DOCS:

iverks, 2025-02-17:
The docstring comments for the [ScatteringFitComponentLobato class](https://github.com/pyxem/pyxem/blob/main/pyxem/components/scattering_fit_component_lobato.py#L29-L111) is on the __init__ function instead of on the class, making them not appear in the [docs.](https://pyxem.readthedocs.io/en/latest/reference/generated/pyxem.components.ScatteringFitComponentLobato.html)

[Here is an example](https://github.com/pyxem/pyxem/blob/main/pyxem/generators/red_intensity_generator1d.py#L31-L283) of a properly docstring-commented class and the [resulting docs.](https://pyxem.readthedocs.io/en/latest/reference/generated/pyxem.generators.ReducedIntensityGenerator1D.html)

CSSFrancis, 2025-02-17:
@iverks Thanks for the issue!  Do you want to make the changes or would you like me to do it?

iverks, 2025-02-18:
I'd prefer if you did it :)

PR #1136 (2025-02-20): Docs: Fix initialization documentation for components.
---
Docs: Fix initialization documentation for components.
---

**Checklist**

- [ ] Docs rendering Properly
- [ ] update the Changelog
- [ ] mark as ready for review

Fixes #1135
closed 2025-02-20 by CSSFrancis